RichiH/modbus_exporter #2: Allow addressing by ID instead of "analog input" etc

RichiH, 2019-04-06:
@mxinden

A default config should probably come with comments to allow people familiar with Modbus to know where to put things in the config and keep a mental map. That being said, addressing by "coil" and other such names seems cumbersome and error-prone. Addressing by ID directly seems better.

Sub-addressing could be done by `id:bit`, ranges could be done explicitly by `id-idN` or implicitly by `id, float32, little endian`.

Gut feeling is that we will develop stronger opinions when using it for some time.

RichiH, 2019-06-18:
@mxinden This is not yet done, correct?

RichiH, 2019-06-18:
Seems it's not, yet.

CC @bastischubert

RichiH, 2019-06-18:
Disregarding that joule as an absolute would be better as we can calculate Wh as a `rate()` from there, I think this would be the better model. Current style:

````
  - name: 'dcim_power_consumption-xxx'
    protocol: 'tcp/ip'
    id: 2
    analogOut:
      - name: power_consumed_watt_hours
        help: 'Power Consumtion in WattHours'
        address: 1226
        metricType: counter
        labels:
          datacenter: 'Halma'
          room: 'xx'
          rack: 'xx'
          fuse: 'xx'
          customer: 'xx'
          billable: 'xx'
        dataType: float32
````

Proposed new style:
````
pre-amble block for the whole target, or exporter instance
  - protocol: 'tcp/ip'
    id: 2 << what is this? Is it unique? Or can it go into pre-amble?

    address: 1226 << this could either be keyed by address, or name. But given that the address is static, it should most likely be keyed by address
        name: 'power_consumed_watt_hours'
        help: 'Power Consumtion in WattHours'
        metricType: counter
        dataType: float32
        labels:
          datacenter: 'xx'
          room: 'xx'
          rack: 'xx'
          fuse: 'xx'
          customer: 'xx'
          billable: 'xx'
````

mxinden, 2019-06-19:
>     `id: 2 << what is this? Is it unique? Or can it go into pre-amble?`

The `id` field or _SlaveID_ is used whenever one talks to a gateway (e.g. TCP/IP gateway) instead of straight to a modbus device. The `id` is used to multiplex the request to the right device behind the gateway. See http://docs.tibbo.com/tmgm/slave-id-based-routing.htm for more details.

---

In regards to the proposed changed format:

- Prometheus needs a way to address the specific module via a `module` http parameter in the configuration file. Thus dropping the `name` property of a module definition makes this addressing impossible. Is there a particular reason why you want to drop it?

---

In regards to changing addressing via ID instead of _analog in/out_ or _digital in/out_  there is a third option: Reading more and more modbus documentations it seems like another specification is _function codes_ see the [Wikipedia article](https://en.wikipedia.org/wiki/Modbus#Format_of_data_of_requests_and_responses_for_main_function_codes).

As of today the specification via  _analog in/out_ or _digital in/out_ seems to work for our generated configuration. For now, given that this is not blocking any functionality, I would suggest postponing this change until we have more feedback by other users in regards to their preference.