RichiH/modbus_exporter #3: Support for 64 bits values and option for BigEndian-LittleEndian

daviddetorres, 2019-11-13:
Hello, 

It would be interesting to add support for the following possible formats:
- 64 bits int
- 64 bits uint
- 64 bits float 

This 64 bit variables are common with dealing with counters (like energy, water, gas meters). 

Also while dealing with different vendors, it is common to have to configure the variable as Big Endian or Little Endian. An option could be added to the config file to specify the kind of codification of the variable. 

If you think any of this ideas would be worth to work in, I could work in a PR.

RichiH, 2019-12-08:
Sorry for the late reply. Very much so, yes. FWIW, floats have the same problems around endianness.

daviddetorres, 2019-12-09:
Thanks for the reply. 

I will work in both 64 bits variables and the "endianity" of the variables applied to all kinds (integers and floats, with and without sign, and 32 to 64 bits). 

I'll leave the current behaviour as default endianity if not configured for backward compatibility and the rest would be needed to be configured.

danielcb, 2019-12-24:
From your talk: "At least I have not seen YOLO endian yet"

Kostal Plenticore Plus (pv inverter) does this. :man_facepalming: 
https://www.photovoltaikforum.com/core/attachment/81082-ba-kostal-interface-modbus-tcp-sunspec-pdf/

Also it uses addresses in the 2-800 range, so I had to add the following (modbus.go:178):
```
...
switch definition.Address / 100000 {
case 0:
    f = c.ReadHoldingRegisters
...
```

I'll help with #6 and, if anyone is interested I also could add an example modbus.yml for the Kostal Plenticore Plus.

daviddetorres, 2019-12-24:
Hi, 

I'm working these days in #6, adding support for Endianness (including Yolo). As you pointed, this is a working mode that I also found in some modbus industrial devices. I expect to have the PR ready to review before end of the year. Reviews on the code and help are welcome!

About the address of the inverter that you commented, note that the register address is created in the following way: 

`
registerAddress = modbusFunctionCode * 10000 + registerPhysicalAddress
`

To be able to read the Generation Energy (decimal address: 577) you have to configure the address 30577 (function code 03: Read Holding Registers).

danielcb, 2019-12-24:
Okay, thanks for pointing that out and thanks for the PR. I'll add the `modbus.yml` example for kostal as soon as your PR is done.

daviddetorres, 2019-12-24:
I finished the PR #6 and marked it ready for review. 

@danielcb: If you want, you can test it for your PV inverter and see if it all works nice. If not, please, let me know so we can fix it before the maintainer merges the PR.

danielcb, 2019-12-24:
Works with my modbus.yml. Thanks!

daviddetorres, 2019-12-24:
That sounds good! 
Merry Xmas!

danielcb, 2019-12-24:
Just in case anyone wants to play around with it and has the same inverter:
https://gist.github.com/danielcb/0d1b23ac209f4312868e937064d6cadf

Merry xmas to you too!

RichiH, 2019-12-27:
Just merged https://github.com/RichiH/modbus_exporter/pull/6 feel free to reopen if needed & thanks!

closed 2019-12-27 by RichiH