RichiH/modbus_exporter #8: Collect modbus.yml for different vendors

RiRa12621, 2020-01-13:
Adding all values to the modbus.yml is a very tedious work.
Doing that once should be enough so it might be good to collect those for vendors/products.
This could be done in a folder and people using the exporter could go through the files, to find out if what they need has already been done.

dssantana, 2020-01-14:
In our particular case, we had all modbus information in a spreadsheet, so we wrote a basic phyton app to create the yml from the spreadsheet. Just a thought

RiRa12621, 2020-01-14:
I fully agree that this can be done and it’s what I will do as well.
I was thinking of this for folks who are not very technically versed and still would like to get visibility on appliances they own like smart heating or power meters.
It’s a lot easier to say “run this docker command and use the file for your device” than “write a script to get the values you need into the right format”

RichiH, 2020-01-15:
If you want to submit anything, I will gladly add it. I am planning to do the same once I have stable configurations

RiRa12621, 2020-01-15:
great, I'll try to find some time to get something ready 👍

rfc1036, 2020-12-08:
I have some configurations that I am happy to share, but I first need to see how issue #9 goes.

RiRa12621, 2023-12-07:
Absolutely can't find the time. Fine to close imho.

closed 2023-12-07 by RiRa12621