RichiH/modbus_exporter #9: Using multiple devices

rfc1036, 2020-01-29:
I do not understand how I can query multiple modbus devices in Prometheus in the best idiomatic way: it would be very helpful to distribute examples of more complex configurations.

I have defined multiple modules in the modbus_exporter configuration, each one detailing the registers of a specific meter that I want to query. In Prometheus now I need to configure multiple RTU gateways, each one containing multiple slave devices from different brands.

What is the correct mix of jobs, targets and sub_targets to do this?

RichiH, 2020-01-29:
While not a direct answer, this modbus_exporter is modeled after https://github.com/prometheus/blackbox_exporter and https://github.com/prometheus/snmp_exporter .

You would usually have one job with multiple targets and sub_targets which have the same configuration in modbus_exporter. A different config would mean a new job.

rfc1036, 2020-11-08:
I have not been able to configure a job with multiple sub_targets because then multiple sub_targets parameters are passed on the modbus_exporter URL and the last one overrides the others.

RichiH, 2020-11-18:
I don't have access to the ModBus environment any more. Maybe @bastischubert can share his config and we can build a reference from that?

Learning by stealing would work best, here.

bastischubert, 2020-12-07:
@rfc1036 currently we are generating a huge pile of yaml within prometheus and the modbus config file. I will share some config examples in the next days

balzreber, 2021-02-15:
Any news on this issue? I have the same problem as @rfc1036. I have a Modbus TCP/RTU converter, and two sensors connected. They are both the same brand and model. How can I get those two into Prometheus as two targets? I tried with sub_targets. But when I list two ['xy', 'xz'], Prometheus just shows two sub_targets as one target. And then writes two sub_targets in one query string. Which, of course, wont work.

fschlich, 2022-10-08:
How about using two prometheus jobs, one with xy and one with yz as sub_target? prometheus iterates over targets, but it can't iterate over other parameters AFAIK

dostalradim, 2022-11-23:
My boss came up with following workarround in `prometheus.yml` and it works well.

```
      - job_name: "inepro-pro1"
        metrics_path: /modbus
        static_configs:
        - targets: ["10.10.2.23:502"]
          labels:
            sub_target: 1
        - targets: ["10.10.2.23:502"]
          labels:
            sub_target: 2
        params:
          module: ["inepro-pro1"]
        relabel_configs:
          - source_labels: [__address__]
            target_label: __param_target
          - source_labels: [sub_target]
            target_label: __param_sub_target
          - target_label: __address__
            replacement: modbus-exporter:9602
```