RichiH/modbus_exporter #11: register address range 0-65535

chakphanu, 2020-04-09:
I need to read the modbus slave from the Deep Sea Generator controller and the register address range reaches up to 64,xxx.
Example register 38656 for reading engine state machine.
If i config modbus.yml to 338656 (add prefix with 3) it's has error below:

>FATA[0000] yaml: unmarshal errors:
>  line 77: cannot unmarshal !!int `338656` into config.RegisterAddr

RichiH, 2020-04-18:
338,656 is way more than 65,535. Are you certain you have the right address?

chakphanu, 2020-04-25:
> 338,656 is way more than 65,535. Are you certain you have the right address?

i mean  in configuration file(yaml).
if i need to read register 38,656,  must add prefix 3 to 38,656.

https://github.com/chakphanu/modbus_exporter/commit/2497ea45a0655f25a79be2f8f8702c92a1a13abf
This is simple hotfix(not smart way).

dshatohin, 2020-04-25:
> if i need to read register 38,656, must add prefix 3 to 38,656.

There is no such register as 338656.
In your case you have register 8656 in Input register space (3).
So it's just 38656.
Check wikipedia page for more details:
https://en.wikipedia.org/wiki/Modbus#Modbus_object_types

closed 2020-12-07 by RichiH