RichiH/modbus_exporter #15: Scaling values

braamvh, 2020-11-06:
Is there a way to scale the values read from Modbus?

In my case I read a number of values off an inverter controller, with all returns being Int or UInt. However, values are scaled where they have decimals.

Example: Battery voltage is returned as 5281 where the real value is 52.81V, while the input voltage is returned as 2303 for 230.3V.

I've looked through the code, but can't see a way to add a scaling factor where the metric is read.

dshatohin, 2020-11-06:
IMO idea behind that is just to give you the real value from modbus register and then transform it by Prometheus.
I'm using PromQL in Grafana exactly the same way - to convert 2300V to 230.0V.

braamvh, 2020-11-06:
Ahhh, I see. Thanks, will try that.

rfc1036, 2020-11-08:
I think that simple scaling like divide by 1000 would still be very useful to better handle instruments which use fixed point arithmetic.

Currently I have to create different Grafana dashboards for different instruments: even if I can normalize the register maps differences to return the same metric with the same name, they still often require different scaling.

SuperQ, 2020-11-09:
Ping @RichiH, IMO this is probably worth doing.

RichiH, 2020-11-18:
I am wondering what the config syntax should be. `factor = n` would seem prudent?

braamvh, 2020-11-18:
`scalefactor = n`, perhaps?  Then just multiply by the value.  So if scalefactor is 0.1 it would divide by 10, and if it is 10 it would multiply by 10. I think that should handle any factors that comes up.

RichiH, 2020-11-18:
I considered `scale` etc, but it might not always be in powers of ten and rather than get into the mathematical weeds thought to keep it simple.

closed 2020-12-07 by RichiH