RichiH/modbus_exporter #17: Re-license from GPLv3 to Apache 2

RichiH, 2020-11-18:
GitHub support just unforked this repository from @lupoDharkael 's original repo.

Given the amount of changes made since forking, that upstream is dead, and that the Prometheus ecosystem runs on Apache 2, it might be prudent to re-license. Ideally, I would like to get confirmation from everyone who committed code in the past to avoid re-implementation efforts.

@chakphanu @mxinden @lupoDharkael @daviddetorres: Would you be OK with this change?

mxinden, 2020-11-18:
Do I understand correctly that you would like to relicense this project under Apache 2?

If so, you have got my approval concerning the code I contributed. I am fine with both Apache as well as the current GPL 3.

RichiH, 2020-11-18:
Yes, that is correct. And thank you.

It's not urgent, or important in the end, I just want to prepare for a future in which Prometheus-team might want to bundle exporters in distribtuions/agents.

daviddetorres, 2020-11-18:
I am OK with the license change.

lupoDharkael, 2020-11-18:
I'm ok with that

RichiH, 2020-12-07:
Looking at https://github.com/RichiH/modbus_exporter/commit/92bcfd3f5befb895d559ff08aebd2b91791077ed I am confident that we can reimplement this patch should @chakphanu object to relicensing.

RichiH, 2020-12-07:
@dshatohin as you committed after this thread started: I presume you're OK with this change as well, correct?

RichiH, 2020-12-07:
Re-licensed.

closed 2020-12-07 by RichiH

dshatohin, 2020-12-07:
@RichiH thank you for mentioning me, even it was just for the sake of formality.
Yes, i don't have any objections 👍