RichiH/modbus_exporter #22: ranges in 10,000 range now cause an error, rejects scraping

rbuckland, 2021-01-08:
The support for large ranges 100,000 has broken the range checking 

```
time="2021-01-08T07:36:15Z" level=error msg="failed to scrape metrics for module 'sma_solar_inverter': metric: 'total_power_watts', address '30770': metric address should be within the range of 10000 - 40000.'1xxxx' for read coil / digital output, '2xxxx' for read discrete inputs / digital input,'3xxxx' read holding registers / analog output, '4xxxx' read input registers / analog input" source="modbus_exporter.go:148"
```

the register in the above example, `30773` is within the range supported. so the check is no longer valid. 

New change - https://github.com/RichiH/modbus_exporter/blob/master/modbus/modbus.go#L182
Previous commit  - https://github.com/RichiH/modbus_exporter/blob/master/modbus/modbus.go#L182

rbuckland, 2021-01-08:
@dshatohin  - see above ; the commit doesn't seem quite right

dshatohin, 2021-01-10:
@rbuckland yep, that was breaking changes which was merged to master.
Check PR for more details:
https://github.com/RichiH/modbus_exporter/pull/19#issuecomment-740206022

rbuckland, 2021-01-11:
I can't use this library as it is (needs support for 32bit values).

As it turns out, my solar inverter supported sunspec modus, so I wrote a prometheus exporter around that library instead.
https://github.com/inosion/prometheus-sunspec-exporter

dshatohin, 2021-01-12:
@rbuckland I don't understand why its not working anymore.
modbus_exporter still able to read any of registers.
Instead of 30773 try to write 300773 in your config file.
Error message is misleading because I forgot to update it.
Sorry for that (will be fixed with #20)

bastischubert, 2021-01-12:
@dshatohin:
What about trying to detect the correct "modulo" for the config (say %1000 Len == 5 and %10000 for Len== 6) 
Old configs would still be valid ;)

dshatohin, 2021-01-12:
@bastischubert yep, I've offered to try this approach some time ago, but [the answer was:](https://github.com/RichiH/modbus_exporter/pull/19#issuecomment-740207369)
> I think that that's rather confusing; as the config is (hopefully) rather static and long-lived, I would prefer to err on the side of easy-to-read not easy-to-write.
But personally I prefer "easy-to-write" approach, so let's try 🔜

vikriusman, 2021-02-19:
hi, is there a temporary solution i can do? I use register 37282 but get an error
`metric address should be within the range of 10000 - 40000.'1xxxx 'for read coil / digital output,' 2xxxx 'for read discrete inputs / digital input,' 3xxxx 'read holding registers / analog output,' 4xxxx 'read input registers / analog input`

PR #23 (2021-03-12): ✨ New easy-to-write register addresses
New address format in `modbus.yml`:
- **first symbol** - modbus function code
- **all other symbols** - modbus register address

Backward compatibility included 🥳

Some tests are fixed (`modbus_test.go`)

Fixes #22
closed 2021-03-12 by bastischubert