RichiH/modbus_exporter #28: Exporter does not connect

vomus, 2021-08-14:
Hi!
I just started to play with modbus_exporter and do not seem to be able to connect to my Modbus device:

```
[serge@veles modbus_exporter]$ ./modbus_exporter 
INFO[0000] Loading configuration file modbus.yml         source="modbus_exporter.go:81"
INFO[0000] telemetry metrics at: :9602                   source="modbus_exporter.go:88"
INFO[0000] Modbus metrics at: :9602                      source="modbus_exporter.go:96"
INFO[0009] got scrape request for module 'safianovo' target '10.161.0.25' and sub_target '10'  source="modbus_exporter.go:130"
ERRO[0009] unable to connect with target 10.161.0.25 via module safianovo  source="modbus_exporter.go:148"
```
although the device itself is fine:
```
[serge@veles modbus_exporter]$ curl 'http://localhost:9602/modbus?target=10.161.0.25&sub_target=10&module=safianovo'
failed to scrape target '10.161.0.25' with module 'safianovo': unable to connect with target 10.161.0.25 via module safianovo

[serge@veles solard]$ mbpoll -q -0 -1 -a 10 -r 122 -t 4:int16 10.161.0.25
-- Polling slave 10...
[122]:  1034
```
Here is my modbus.yml:
```
modules:
  - name: "safianovo"
    protocol: 'tcp/ip'
    metrics:
      - name: "battery_voltage"
        help: "Battery voltage"
        labels:
          phase: "1"
        address: 122
        dataType: int16
        endianness: little
        metricType: gauge
```
What am I doing wrong?

dshatohin, 2021-08-14:
Hi, try to specify address with address range (1, 2, 3 or 4):

```yaml
modules:
  - name: "safianovo"
    protocol: 'tcp/ip'
    metrics:
      - name: "battery_voltage"
        help: "Battery voltage"
        labels:
          phase: "1"
        address: 400122
        dataType: int16
        endianness: little
        metricType: gauge
```

vomus, 2021-08-14:
Nop. Does not work either

RichiH, 2021-08-14:
Can you reach this target with other Modbus/TCP tools?

Sent by mobile; please excuse my brevity.

vomus, 2021-08-14:
> Can you reach this target with other Modbus/TCP tools? Sent by mobile; please excuse my brevity.

Of course. With mbpoll. I just gave the example in my original post:

```
[serge@veles modbus_exporter]$ mbpoll -q -0 -1 -a 10 -r 122 -t 4:int16 10.161.0.25
-- Polling slave 10...
[122]:  1025
```

vomus, 2021-08-14:
I see a timeout parameter in the code. Can it be set? What's the default value?

vomus, 2021-08-15:
Although, I am not at all a go programmer, I went to dig in "github.com/goburrow/modbus" that you are using and cooked up a little piece of code that could connect to my Modbus device

```
[serge@veles mgo]$ go run mgo.go 
test: 2021/08/15 07:42:40 modbus: sending 00 01 00 00 00 06 0a 03 00 7a 00 02
test: 2021/08/15 07:42:40 modbus: received 00 01 00 00 00 07 0a 03 04 04 0a 03 0e
[4 10 3 14]
```

```
func main() {
	handler := modbus.NewTCPClientHandler("10.161.0.25:502")
	handler.Timeout = 10 * time.Second
	handler.SlaveId = 10
	handler.Logger = log.New(os.Stdout, "test: ", log.LstdFlags)
	err := handler.Connect()
	defer handler.Close()

	client := modbus.NewClient(handler)
	results, err := client.ReadHoldingRegisters(122, 2)
	if err != nil {
		fmt.Println("Error %v", err.Error())
	}

	fmt.Println(results)
}

```

RichiH, 2021-08-20:
Can you try a manual timeout in modbus exporter directly? From a quick glance we have a default timeout of 0. I would be wary of adding too much time in general, so maybe we need to add a per-module config option.

tward333, 2022-02-17:
I had this same issue and was able to resolve it by specifying a port in my target (default 502). Does this work for you?
` curl 'http://localhost:9602/modbus?target=10.161.0.25:502&sub_target=10&module=safianovo'`

Found this out after trying to replicate the included fakeserver configuration and it too was failing. `target=127.0.0.1` didn't work but `target=127.0.0.1:1502` does

PR #36 (2022-10-09): README: mention that sub_module is a required parameter and target ne…
…eds to include the port (fixes: #28)

Also add a helpful hint to modbus.yml how to turn "two registers wide" into a datatype.

This is info that would have helped me a lot when starting to use modbus_exporter. I hope this PR can make it more visible.
commit 0cb1b1b7 (2022-10-09): README: mention that sub_module is a required parameter and target needs to include the port (fixes: #28)

Also add a helpful hint to modbus.yml how to turn "two registers wide"
into a datatype.
closed 2022-10-09 by bastischubert