RichiH/modbus_exporter #34: Catch and expose well-defined error/information values

RichiH, 2022-08-23:
Some devices send special and locally-well-defined numeric values to signal state, e.g. "Resource busy".

The initial plan was to hide those values and print errors to STDOUT, see https://github.com/RichiH/modbus_exporter/pull/32

This is not ideal as it forces people to look at the logs, hides information from PromQL, and makes these situations generally invisible/harder to debug.

Current thinking is to introduce a gauge which exposes this value instead.

So metric `foo` will dynamically get a `foo_caught = 1234` where `1234` is the magic value. While OpenMetrics StateSet would allow for better and cleaner mapping, it would multiply the metrics exposed.

Two open questions:
* Should `foo` go away, be set to zero, or retain its old value? Going away for the time being seems cleanest.
* Should `foo_caught` go away once the special state is gone, be set to zero, or be set to zero and then go away? Going away seems cleanest yet again.

CC @SuperQ for thoughts.

RichiH, 2023-03-14:
@bastischubert @DaAwesomeP @SuperQ thoughts?

DaAwesomeP, 2023-03-14:
IMO (and I am definitely and absolutely not a definitive source, just my opinion):
 - `foo` should go away when there is not a current value to expose, so it should go away on error
 - However with `foo_caught` I am more of the opinion that there should be a `0` state to signify "OK" since it is very possible a user would forget to implement alerts or visualization if it is not present initially. I think also that "no errors caught/operating normally" is definitely a valid state separate from the actual value of the sensor/device, which would be in `foo`.
 - Maybe instead of calling it `foo_caught` (which implies that it should disappear if nothing is caught) it should instead be `foo_error` which can continue to exist to say "no current error."

Are there other exporters we can look at that have similar metrics or have solved a similar issue? Not saying we should blindly copy it but it would be good to see other examples.

I should also note that I don't think I have any devices on-hand that support these error codes.

DaAwesomeP, 2023-03-14:
>Signalling OK state would mean doubling the metric count. I am not against it, yet still apprehensive.

Yeah...understandable.

> Or maybe we could expose this only for metrics which have a special handler defined?

Yeah, that makes sense; then you're only adding metrics where it would actually work. And I imagine that most devices don't support this exceptions of every single address anyway.

In my case I am only monitoring simple I/O devices so I would almost never encounter this feature and would leave it off.

bastischubert, 2026-10-16:
Picking this up. Proposal building on the discussion above:
- opt-in per metric via `specialValues: [{value, meaning}]`; only metrics with a handler can produce extra series
- on match: `foo` is omitted from that scrape (other metrics unaffected)
- instead of a per-metric `foo_caught`, one fixed family: `modbus_special_value_caught{reading="foo", address="…", meaning="…"} <raw value>`, so one alert rule and one dashboard panel cover all devices, with the same series count as `foo_caught`
- once the state clears, the series goes away; alerting on its presence covers the "0 state" concern

I'll rework #32 accordingly unless there are objections.

RichiH, 2026-10-16:
Re-reading the thread, I am wondering if we can simply have `foo` for the value and `foo_last_udpated` to catch when it was updated.

It's also fine to experiment with the above, I feel.

bastischubert, 2026-10-17:
The approach from the proposal above shipped in [0.5.0](https://github.com/RichiH/modbus_exporter/releases/tag/v0.5.0) via #32: `specialValues` per metric, exposed as `modbus_special_value_caught{reading, address, meaning}`.

I looked into `foo` + `foo_last_updated` and decided against it:

- **2x metrics:** every reading needs a companion `_last_updated` series, so each configured metric doubles the series count, all the time and not just while a special value is present. `modbus_special_value_caught` only exists while a device actually reports a special value, and only for metrics with `specialValues` configured.
- **Complicated alert rules:** staleness alerts would look like `time() - foo_last_updated > 300` for every single reading, with thresholds tuned to each scrape interval, and no way to see *why* a value went stale. With one fixed family, a single rule covers every device and module, and the `meaning` label tells you what the device reported:
  ```yaml
  - alert: ModbusSpecialValueCaught
    expr: modbus_special_value_caught
    annotations:
      summary: "{{ $labels.reading }} on {{ $labels.instance }}: {{ $labels.meaning }}"
  ```
- **State:** keeping the last good `foo` would also require a per-target cache in the exporter, which is stateless per scrape today. That cache would be lost on restart.

Closing this as implemented by #32. If anyone runs into cases the current approach doesn't cover, feedback is very welcome, either here or in a new issue.

closed 2026-10-17 by bastischubert