RichiH/modbus_exporter #38: Inverter Huawei SUN2000, scrape i/o timeout

maleksan, 2023-02-02:
Inverter Huawei SUN2000, scrape i/o timeout.

Trying to read modbus(Inverter Huawei SUN2000) "Active Power" data, but there is such error:
```code
time="2023-02-02T14:14:56Z" level=info msg="got scrape request for module 'fake' target '<modbus_ip_adress>:1502' and sub_target '2'" source="modbus_exporter.go:81"
time="2023-02-02T14:15:01Z" level=error msg="failed to scrape metrics for module 'fake': metric 'active_power', address '32080': read tcp ****:33534-><<modbus_ip_adress>>:1502: i/o timeout" source="modbus_exporter.go:96"
```

modbus.yml cofig:
```yaml
modules:
  # Module name, needs to be passed as parameter by Prometheus.
  - name: 'fake'
    protocol: 'tcp/ip'
    metrics:
      # Name of the metric.
      - name: 'active_power'
        help: 'active power'
        address: 32080
        dataType: int32
        metricType: gauge
```

Modbus unitid is 2.

Url:
```code
http://localhost:9602/modbus?target=<modbus_ip_adress>:1502&module=fake&sub_target=2
```

Modbus unit port is open:
```bash
nc -zv <modbus_ip_adress> 1502
Connection to <modbus_ip_adress> 1502 port [tcp/*] succeeded!
```

Huawei register definitions documentation says the following:
[image]

What is wrong in my configuration?
In the meantime, I can read Modbus data with the Node-red module - https://flows.nodered.org/node/node-red-contrib-modbus

maleksan, 2023-02-07:
I managed to solve this problem in the Telegraf.
Such Modbus timeout issue is fixed in Telegraf with configuration "workarounds":
https://github.com/influxdata/telegraf/blob/release-1.25/plugins/inputs/modbus/README.md#workarounds

In my case, by setting the `pause_after_connect` option to 2s, I no longer have timeout errors.
Also options such as `pause_between_requests`, `close_connection_after_gather` and `one_request_per_field` would be useful in case of slow modbus devices.
Is it possible to implement such options in Prometheus modbus exporter?

florianhuse1988, 2024-01-09:
@maleksan 
I tried today to access an energy meter (Janitza) and struggled with the same issue.

Solution:
Adjust the address to **3**32080. --> the additional 3 is the missing function code (described in modbus.yml example).

PR #65 (2024-03-06): Add workaround to sleep after connect
Fixes #38
closed 2024-03-06 by bastischubert