RichiH/modbus_exporter #61: Wrong values when reading SMA inverter

LukeLR, 2023-09-16:
Hi, thanks for this great project. I want to create an SMA inverter config, but I'm failing to read registers.

[SMA Modbus TCP documentation](https://files.sma.de/downloads/PARAMETER-HTML_STPxx-3SE-40_30109R_V11.zip) taken from [here](https://www.sma.de/produkte/hybrid-wechselrichter/sunny-tripower-smart-energy) lists e.g. register 31435 as a signed int32 value. Reading this register using pymodbus or qmodmaster gives two 16bit values 0, 1716. However, reading the register using modbus_exporter always gives me -1. Am I doing something wrong? Here's my config:

```
modules:

    # Module name, needs to be passed as parameter by Prometheus.
  - name: "SMA"
    protocol: 'tcp/ip'
    metrics:
        # Name of the metric.
      - name: "power_consumption_total"
        # Help text of the metric.
        help: "represents the overall power consumption by phase"
        # Labels to be added to the time series.
        labels:
          phase: "1"
        # Register address.
        # The first digit of the address is the function code
        # Supported codes are: 1, 2, 3, 4
        address: 31435
        # Datatypes allowed: bool, int16, int32, int64, uint16, uint32, uint64,
        #   float16, float32, float64
        # One register holds 16 bits.
        dataType: int32
        # Endianness allowed: big, little, mixed, yolo
        # Optional. If not defined: big.
        endianness: big
        # Prometheus metric type: https://prometheus.io/docs/concepts/metric_types/.
        metricType: gauge
        # Factor can be specified to represent metric value.
        # Examples: 1, 2, 1.543, 0.01 etc
        # Factor is multiplied with the scraped value to produce the metric value
        # Optional.
```

Any ideas appreciated!

bastischubert, 2023-09-18:
Hi Luke, 
have you tried to capture the data that's going over the wire (tcpdump/wireshark) and compared if the device was "asked" for the same data?

LukeLR, 2023-09-18:
Good point, I also thought about that but somehow forgot it. I'll try that :)

LukeLR, 2023-09-25:
> Thanks a lot for the hint. By capturing my traffic using Wireshark, I found out that the first digit in the SMA register IDs is not the function code, e.g. although 3 is the correct function code for register 31435, the function code is not yet included in the register ID. It works in qmodmaster and pymodbus, because they keep the register ID as is. However, modbus_exporter strips the leading digit from the ID as the function code, so it queries register 1435, while SMA expects you to actually query register 31435, although 3 is also the function code. Therefore, you have to prepend another 3 to the register address in modbus_exporter to get the right value. And I'm pretty sure I tried that, but probably I didn't :)
>
> Sorry for the noise, but talking about that issue really helped!
>
> Closes #61

And for everyone who wants to export SMA metrics to prometheus using modbus as well, I've created a complete SMA config for modbus exporter and published it here: https://codeberg.org/LukeLR/sma_modbus_prometheus_exporter

bastischubert, 2023-09-25:
Hi Luke,
generally you'd need to prefix the register adress with the code (see example file -> https://github.com/RichiH/modbus_exporter/blob/main/modbus.yml#L15)

closed 2023-09-25 by bastischubert