RichiH/modbus_exporter #78: scraping strings

stefangweichinger, 2024-11-07:
Started to use this yesterday, so pls forgive me asking newbie questions.

I read data from a device providing string values like software version and serial number.

From your example `modbus.yml` I don't see that a datatype "string" is allowed. Is that true?

(how) can I export these registers with this exporter? thanks in advance

omnistat, 2024-11-07:
Currently, the exporter [only supports numeric data types](https://github.com/RichiH/modbus_exporter/blob/62be469666f517cd5801a5bb4f1b8f31e4a70353/config/config.go#L88-L100).     This limitation is because [Prometheus stores sample values as float64 only](https://prometheus.io/docs/concepts/data_model/#samples:~:text=and%20labels.-,Samples,-Samples%20form%20the).

Some exporters, like the SNMP exporter, allow encoding of certain string values into the labels of a Prometheus metric, but this isn't supported by this exporter. 

If you want to be able to read string data, you may be better off using a monitoring setup with a database that supports strings, such as Telegraf with Influx or TimescaleDB.

SuperQ, 2024-11-08:
Supporting info metrics (string labels) is something that could be implemented here. Someone would need to contribute the feature.