RichiH/modbus_exporter #81: modbus register address parcing failed  0

Khao-thin, 2025-04-29:
This error is reported when executing according to the tutorial. What is the cause? How to solve it?
The following is my modbus.yml

modules:
    # Module name, needs to be passed as parameter by Prometheus.
  - name: "fake"
    protocol: 'tcp/ip'
    slave_id: 1
    metrics:
        # Name of the metric.
      - name: "power_consumption_total"
        # Help text of the metric.
        help: "represents the overall power consumption by phase"
        # Labels to be added to the time series.
        labels:
          phase: "1"
        # Register address.
        # The first digit of the address is the function code
        # Supported codes are: 1, 2, 3, 4
        address: 1
        # Datatypes allowed: bool, int16, int32, int64, uint16, uint32, uint64,
        #   float16, float32, float64
        # One register holds 16 bits.
        dataType: int16
        # Endianness allowed: big, little, mixed, yolo
        # Optional. If not defined: big.
        endianness: big
        # Prometheus metric type: https://prometheus.io/docs/concepts/metric_types/.
        metricType: counter
        # Factor can be specified to represent metric value.
        # Examples: 1, 2, 1.543, 0.01 etc
        # Factor is multiplied with the scraped value to produce the metric value
        # Optional.
        factor: 3.1415926535

Below is a screenshot of my Modbus Slave
[image]
The following is my error screenshot
[image]
I tried to change the address to 40002 and got the following error:
[image]

closed 2025-05-14 by Khao-thin

Atulsingh92, 2025-09-25:
But why and how???

sc-alex, 2026-09-29:
@Atulsingh92 By RTFM probably :)

On this exporter in its current form the function is encoded into the `address` variable as a prepending integer number.

The address posted has only one digit

> ```yaml
> metrics:
> […]
>   address: 1
> ```

and that's being decoded to be the function [01 \(0x01\) Read Coils \(p.11\)](https://www.modbus.org/file/secure/modbusprotocolspecification.pdf), but the register address to query is still missing then.